wpilibsuite/allwpilib #2345: CircularBuffer (cpp) is unsafe

Oblarg, 2020-02-06:
The C++ `CircularBuffer` class returns `0` from a method that returns a reference-type.  Attempting to actually do anything with this return value would be undefined behavior.

https://github.com/wpilibsuite/allwpilib/pull/2344 changes this to return an actual member, which at least will continue to exist, but this is still wrong as a reassignment of this member will result in incorrect behavior in the future.

calcmogul, 2020-02-06:
There isn't anything to fix here. Since it's styled as an STL container, it should have the same semantics. std::vector::back() is UB if called on an empty container.

https://en.cppreference.com/w/cpp/container/vector/back

calcmogul, 2020-02-06:
We could add a comment stating it's UB like the std::vector docs do.

calcmogul, 2021-07-11:
circular_buffer now has docs on functions like pop_back() stating it's UB to call on an empty buffer.

closed 2021-07-11 by calcmogul