wpilibsuite/allwpilib #2353: [Feature Request] InitDouble, InitBoolean, etc in Preferences class

reteps, 2020-02-10:
Our team, The RatPack, is currently using preferences to read important values for the robot. When we start our code, we initialize all preferences values by first checking if they exist, then performing a put call if they do not.
```cpp
// robot.h
frc::Preferences& prefs = *frc::Preferences::GetInstance();
// robot.cpp RobotInit
if (!prefs.ContainsKey(name)) {
    prefs.PutDouble(name, val);
}
// robot.cpp RobotPeriodic
double importantRobotValue = prefs.GetDouble(name, 0);
```

It would be great if this logic could be inside the Preferences class, so that our code inside RobotInit would look like:
```
prefs.InitDouble(name, val)
```
This does not look like a very hard fix, and I think it would be useful to teams who are accidentally overwriting their saved preference values.

https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/src/main/native/cpp/Preferences.cpp

ThadHouse, 2020-02-10:
We actually already have support for doing exactly this in NetworkTables itself (that's what the SetDefault functions do) but I agree adding them directly to preferences could be helpful.

reteps, 2020-02-10:
I think adding them directly would be helpful just so it is less confusing for newer students. 
> Why are we using the table "Preferences" here but later reading from the "Preferences class"?
Having this seperation of logic can be confusing.
```cpp
frc::Preferences& prefs = *frc::Preferences::GetInstance();
NetworkTable.GetTable("Preferences").SetDefaultDouble(name, val);
prefs.GetDouble(name, 0);
```

closed 2020-04-02 by PeterJohnson

reteps, 2020-04-02:
awesome!

PR #2443 (2020-04-02): [wpilib] Add init methods to Preferences
Closes #2353