wpilibsuite/allwpilib #2356: close() doesn't do anything in RobotBase

retrodaredevil, 2020-02-11:
RobotBase inherits from AutoCloseable and its `close()` method does nothing and is not called. Is this intentional?

I know we have `endCompetition()`, but I think it would be more intuitive to have `close()` do something or remove it entirely. Maybe we should have `endCompetition()` call close?

This affects the Java API, I'm not sure about the C++ API.

ThadHouse, 2020-02-11:
It's empty because we don't want to require users to implement it, but want users to be able to override it if necessary.

Currently, robot close never shuts down anyway, so it would never get called, but in sim it's something we might look into getting working.

Starlight220, 2020-04-06:
Is this still relevant/active? If not, should be closed.

closed 2020-04-06 by retrodaredevil