wpilibsuite/allwpilib #2357: Minor documentation error for SelectCommand

tervay, 2020-02-13:
The `SelectCommand` docs (https://docs.wpilib.org/en/latest/docs/software/commandbased/convenience-features.html#selectcommand) have a minor mistake in how the `Map` object is constructed. The example code won't compile.

Example Code:

```java
    // The enum used as keys for selecting the command to run.
    private enum CommandSelector {
      ONE, TWO, THREE
    }
  
    // An example selector method for the selectcommand.  Returns the selector that will select
    // which command to run.  Can base this choice on logical conditions evaluated at runtime.
    private CommandSelector select() {
      return CommandSelector.ONE;
    }
  
    // An example selectcommand.  Will select from the three commands based on the value returned
    // by the selector method at runtime.  Note that selectcommand works on Object(), so the
    // selector does not have to be an enum; it could be any desired type (string, integer,
    // boolean, double...)
    private final Command m_exampleSelectCommand =
        new SelectCommand(
            // Maps selector values to commands
            Map.ofEntries(
                entry(CommandSelector.ONE, new PrintCommand("Command one was selected!")),
                entry(CommandSelector.TWO, new PrintCommand("Command two was selected!")),
                entry(CommandSelector.THREE, new PrintCommand("Command three was selected!"))
            ),
            this::select
        );
```

The fixed code should replace `Map.ofEntries(entry(...), entry(...), entry(...))` with `Map.ofEntries(Map.entry(...), Map.entry(...), Map.entry(...))`.

Would fix but not sure where to find in docs.

Oblarg, 2020-02-13:
It does compile; you need to statically import `Map.entry`.

tervay, 2020-02-14:
Then perhaps that should be stated in the example. It's not clear that it is using a static import.

AustinShalit, 2020-02-18:
This example should be updated to not use static imports

PR #2442 (2020-03-22): Remove static import usage in SelectCommand example
Closes #2357
closed 2020-03-22 by PeterJohnson