wpilibsuite/allwpilib #2363: Missing CTRE_PhoenixCCI running Simulator on MacOS

randomstring, 2020-02-15:
I'm trying to run the simulator on my Mac and am stuck and am getting the following error. The GUI comes up, briefly, and then crashes. Obviously I need CTRE_PhoenixCCI installed, but am stumped on how and where. 

`HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
windowScale = 2.000000e+00Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
********** Robot program starting **********
NT: server: client CONNECTED: 127.0.0.1 port 55481
java.lang.UnsatisfiedLinkError: no CTRE_PhoenixCCI in java.library.path: [/Users/dole/GitHub/Sonic-Squirrels/2020-RobotCode/build/tmp/jniExtractDir]
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2660)
        at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:829)
        at java.base/java.lang.System.loadLibrary(System.java:1867)
        at com.ctre.phoenix.CTREJNIWrapper.<clinit>(CTREJNIWrapper.java:9)
        at com.ctre.phoenix.motorcontrol.can.BaseMotorController.<init>(BaseMotorController.java:60)
        at com.ctre.phoenix.motorcontrol.can.BaseTalon.<init>(BaseTalon.java:31)
        at com.ctre.phoenix.motorcontrol.can.TalonFX.<init>(TalonFX.java:27)
        at com.ctre.phoenix.motorcontrol.can.WPI_TalonFX.<init>(WPI_TalonFX.java:62)
        at frc.robot.subsystems.driveSubsystem.<init>(driveSubsystem.java:39)
        at frc.robot.RobotContainer.<init>(RobotContainer.java:50)
        at frc.robot.Robot.robotInit(Robot.java:29)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:64)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:276)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:329)
        at java.base/java.lang.Thread.run(Thread.java:834)`

sciencewhiz, 2020-02-15:
I don't think that CTRE supports simulation on mac, only on windows/linux. 
https://phoenix-documentation.readthedocs.io/en/latest/blog/blog-week4.html#new-phoenix-api-v15-3

virtuald, 2020-02-15:
Correct. If you want OSX support for CTRE products, I recommend sending them an email.

closed 2020-02-15 by virtuald

randomstring, 2020-02-15:
That would explain it.

Am I missing something or does this pretty much make the simulator unusable on Mac unless you don't use any CTRE products.

ThadHouse, 2020-02-15:
You can use the RobotBase.isReal and RobotBase.isSimulation functions to only conditionally initialize CTRE code, allowing you to simulate the other portions of your robot.