wpilibsuite/allwpilib #2368: Calling stop() on a notifier doesn't actually stop it.

notmattlythgoe, 2020-02-18:
When calling stop() on a Notifier the notifier doesn't actually stop running.

ThadHouse, 2020-02-18:
Can you post the code you're trying to use? I can't reproduce this locally with WPILib 2020.2.2

notmattlythgoe, 2020-02-18:
https://github.com/TripleHelixProgramming/InfiniteRecharge/blob/master/src/main/java/frc/robot/drivetrain/commands/TurnToAngle.java

notmattlythgoe, 2020-02-18:
Also happening here:

https://github.com/TripleHelixProgramming/HelixUtilities/blob/master/src/main/java/com/team2363/commands/HelixFollower.java

ThadHouse, 2020-02-18:
I wonder if this is a command issue with end() not working correctly. Can you verify that end() is actually getting called?

notmattlythgoe, 2020-02-18:
It would have to be an issue with end and interrupt not working correctly. The HelixFollower is specifically showing up when the autonomous command is getting cancelled before the path is finished.

notmattlythgoe, 2020-02-18:
This team wrapped HelixFollower in the new command structure and is experiencing the same issues.

https://github.com/619Code/Robot2020CommandBased/blob/master/src/main/java/frc/robot/helpers/HelixFollowerCommandWrapper.java

ThadHouse, 2020-02-18:
I just tried doing exactly what those commands are doing, and it's working fine. end() is getting called, and the notifier properly stops.

We didn't change anything in 2020.2.2, but I noticed that project is still 2020.1.2. Can you try updating?

notmattlythgoe, 2020-02-18:
What about the turn to angle example? That's on a project that is on 2020.2.2?

ThadHouse, 2020-02-18:
Ah I didn't see that project was on 2.2. I don't have Sparks so I can't get your exact program to run, so I'm trying to rip things out. But just a stock command starting a notifier in initialize() and stopping it in end() works correctly in multiple test projects. So trying to think of other ways that would break.

notmattlythgoe, 2020-02-18:
What about doing it from command groups?

ThadHouse, 2020-02-18:
Just tried that, and that worked without issue.

ThadHouse, 2020-02-18:
Hmm. By doing some more interesting things in a CommandGroup it looks like I can reproduce this. Let me try out some things.

notmattlythgoe, 2020-02-18:
The issue seemed to first appear when transitioning out of auto when the command group wasn't finished

ThadHouse, 2020-02-18:
I found the issue. It was caused by a change in the way we did locking previously. It's actually a race, but by having the notifier period as high as your code its easier to see. It wasn't detectable when I was only running at 20ms periods.

notmattlythgoe, 2020-02-18:
Awesome!

PR #2370 (2020-02-19): Fix notifier race happening with very fast notifiers
What would happen is the Stop() call would happen between the notifier loop being triggered and calling UpdateAlarm(). This would cause the Update to overwrite the stop.

Closes #2368
closed 2020-02-19 by PeterJohnson