wpilibsuite/allwpilib #2369: Timer class resets on start if it was already running

pauloalbert, 2020-02-18:
The timer class doesn't save the accumulated time if you start the timer without calling stop beforehand, and resets `m_startTime` anyways.
I was wondering if this is indeed an intended feature, because from my experience it wasn't intuitive for us and it caused us confusion why the timer was staying at 0.

Oblarg, 2020-02-18:
I agree that this is confusing; `start()` should be a no-op if the timer is already running.

PeterJohnson, 2020-02-19:
We don't want to fix this during the season, as it's a silent change to behavior.  We'll consider this for 2021.

Starlight220, 2020-03-16:
When this is changed, make sure that the documentation includes the info that running start() on a running timer doesn't do anything.

PR #2434 (2020-03-21): Timer.start(): Match C++ behavior in Java
Closes #2369
closed 2020-03-21 by PeterJohnson