wpilibsuite/allwpilib #2374: Test mode could potentially be triggered during a match

virtuald, 2020-02-19:
IterativeRobotBase contains a race that could cause a robot to go into test mode instead of the directed mode. I think the only way it could happen is the transition between autonomous -> disabled.

There are three cases that check the mode -- `IsDisabled` followed by `IsAutonomous` followed by `IsOperatorControl`, followed by an else statement. Each variable is retrieved separately, no lock.

If during the first check the robot was in autonomous, but between that check and the `IsAutonomous` check the robot dropped to disabled, it would fall through to the else and put the robot momentarily into test mode. It's almost certainly exceedingly rare, but I bet it does happen sometimes.

The ideal fix for this would retrieve a control word at the beginning of LoopFunc, and use the same structure for all of the comparisons.

retrodaredevil, 2020-02-19:
I think that this bug affects more than just IterativeRobotBase. This could affect any program that makes calls to the DriverStation class.

(I'm looking at the Java code)
The way this bug would actually happen would be if the `System.currentTimeMillis()` call increments on the same iteration and it causes the `now - m_lastControlWorldUpdate > 50` to be true.

It's very rare, but I think a better solution would be to actually fix the bug in the DriverStation class so that any code that uses DriverStation would always work as expected.

Also, I believe that your logic is slightly incorrect: `isDisabled()` can return true while `isAutonomous()` returns true (a robot can be disabled, but can also check what the current mode is). So the worst thing that would actually happen would be that code is that code meant for an enabled robot is ran while it's disabled, which is possible if the robot is disabled while in the middle of an iteration.

I think this problem needs to be fixed, but I don't think "test" code would be ran, I'm pretty sure the outcome would be in your example that autonomous code is ran for an extra iteration.

Starlight220, 2020-03-22:
@virtuald , has this actually happened or is it an algorithmic "weakness" found by looking at source code? If the latter, I think that the consequences of this weakness should be explored before fixing. If the only consequence is another iteration of auto, I think there is little reason to work on this.

virtuald, 2020-03-22:
I have not personally tried to cause this to happen, and I imagine if it did happen it's unlikely a team would be able to track it down.

This is definitely a 'code smell' issue to me because it's difficult to prove that the code is obviously correct, and if the error did occur it would be immensely difficult for a team to figure out what occurred. It would be easy to fix this so that it's obviously correct, so we should do that.

Starlight220, 2020-04-19:
As it seems to me, it will be relatively easy to fix the IterativeRobotBase issue. However, fixing the issue @retrodaredevil mentioned:

> The way this bug would actually happen would be if the `System.currentTimeMillis()` call increments on the same iteration and it causes the `now - m_lastControlWorldUpdate > 50` to be true.
>
> It's [...]

would be much more difficult. I will write a PR for the IterativeRobotBase issue, as I can't think of a way to fix the timing issue.

Starlight220, 2020-04-19:
> Also, I believe that your logic is slightly incorrect: `isDisabled()` can return true while `isAutonomous()` returns true (a robot can be disabled, but can also check what the current mode is). So [...]

I think you might be mistaken, as `disabled` is the first branch of the `else-if` tree. 

What I think I will do is explicitly checking for test mode, then if test mode check fails, I think an exception should be thrown / error be printed stating that the robot is in an illegal mode.

virtuald, 2020-04-19:
It really should be retrieving a single ControlWord, and then performing operations based on that control word, instead of calling multiple functions that could race. Given a control word, I don't believe there are any combinations that lead to an undefined state, so none of your options would be necessary.

retrodaredevil, 2020-04-19:
I think the control word needs to be cached as soon as possible. If you cache the control word, you shouldn't have any problems. Right now, I think the control word is mutable, which means that there are possibly some race conditions.

Also, I'll simplify the check that happens:

```
// we're in teleop
if(isDisabled() { // 

} 

else if (isAutonomous() {

} 

else if(!isAutonomous() && !isTest()) { // this is what isTeleop() actually checks

} else { // assume in test mode

}
```
Let's assume the following scenario: Robot is in teleop, then robot becomes disabled. What's going to happen is that isDisabled goes from false to true, and isAutonomous and isTest both stay false. So let's say the first check if isDisabled is false, then the robot becomes disables right after that check. When that happens, isAutonomous and isTest are both still false, so teleop code is run for an iteration when the robot is actually disabled.

I don't know exactly what happens during the transition from autonomous to teleop, but let's say that the robot goes from autonomous enabled to teleop disabled. Once again, the first check to isDisabled is false, then isAutonomous and isTest are both false, running teleop code before we want to.

It will never just fall through to test code because the control word only had three important booleans that represent the mode: disabled, autonomous, and test. If the isDisabled check is false, but becomes true, then enabled code will run for whatever mode it is in.

And @virtuald beat me to the single control word thing. I definitely agree with that. Just make sure that you watch out for mutability.

Starlight220, 2020-04-19:
I made the PR before I saw your replies, and didn't think about the control word. How do I implement that? I understand that the mode should be queried at the beginning. Do I simply use a mode enum value, or do I use a ControlWord class or something?

retrodaredevil, 2020-04-19:
I recommend that in RobotBase, you create a ControlWord instance as a member field, then first thing in the loop, call `HAL.getControlWord(m_controlWord)`. So something like this:

```
class RobotBase {
    private final ControlWord m_controlWord = new ControlWord();
    void update() {
        HAL.getControlWord(m_controlWord);
        if (isDisabled() {} ... {} ....
    }
}
``

Starlight220, 2020-04-19:
I added the control word, thoughts?

ThadHouse, 2021-08-06:
Technically, https://github.com/wpilibsuite/allwpilib/pull/3499 fixes this bug. It still does the multiple reads, but theres no else if statement to fall through.

closed 2021-08-06 by PeterJohnson