wpilibsuite/allwpilib #2376: Overridden periodic method in PIDSubsystem causes super() to not compile properly

Phantom2x5, 2020-02-19:
When overriding the periodic() method within a PIDSubsystem, the super() method that creates the new PIDController fails to instantiate. Other separate variables and methods within the PIDSubsystem work, including the override of the periodic() method. However, the PIDController, will not work at all because its instantiation fails to occur due to the periodic() method being overridden.

Oblarg, 2020-02-19:
Nothing is instantiated in the `periodic()` method of `PIDSubsystem`.

If you override `periodic()` and fail to call `super()`, then yes, the PID control will not work.  This is normal and expected behavior.

sciencewhiz, 2020-02-19:
The old framework uses shadow methods so that users can't override something in a way that breaks.

Phantom2x5, 2020-02-19:
That makes sense. This is only my second year doing this, so I wasn't quite sure about it.

closed 2020-02-19 by Phantom2x5

Oblarg, 2020-02-19:
> The old framework uses shadow methods so that users can't override something in a way that breaks.

Yeah, but it's understandable that a user might want to replace the periodic method with their own implementation.  It's not so hard to call `super()`, and it's not surprising that it breaks if you don't.