wpilibsuite/allwpilib #2379: Refactor Tracer functionality out of Watchdog

calcmogul, 2020-02-20:
Watchdog is intended for calling a callback upon timeout. The execution tracing functionality, while useful, isn't related to that main purpose.

If we split that into a separate Tracer class, we could make IterativeRobotBase pass a lambda callback to Watchdog that prints the Tracer contents. We could expose functions in IterativeRobotBase for the Tracer so users can add more fine-grained epochs.

Starlight220, 2020-03-24:
In "execution tracing functionality" do you mean the epoch and time recording?

Starlight220, 2020-03-24:
Should the Watchdog class have an internal Tracer with wrapper functions?

calcmogul, 2020-03-24:
Yes. Watchdog should maintain the same API.

Starlight220, 2020-03-24:
Though I am adding one thing to the API : a printEpochsOnTimout flag. You are welcome to help me come up with a better name.

calcmogul, 2020-03-24:
Watchdog already does that implicitly though.

Starlight220, 2020-03-24:
No, it doesn't, IterativeRobotBase handles it. Therefore I suggest adding the function and using it in IterativeRobotBase instead of the workaround that is currently used.

Starlight220, 2020-03-24:
I am also setting the output to be a single block instead of multiple print statements to make sure that the output is not divided by any other output.

calcmogul, 2020-03-24:
You don't want epochs printing on timeout because they are incomplete. You have to wait for whatever the user was doing to finish before printing the epochs. That's why the user has to handle it.

Starlight220, 2020-03-24:
Since this is a major refactor to the Watchdog class, should I squash #2413 / #2178 into this refactor as well?

Starlight220, 2020-03-24:
And do you think Watchdog should inherit from Tracer?

calcmogul, 2020-03-24:
No, keep #2413 separate. We can merge that first.

No, Watchdog should not inherit from Tracer. Composition is generally superior to inheritance.

Starlight220, 2020-03-24:
About #2413, I am leaving the Watchdog output as is (to System.out), but I am outputting the Tracer output to the DS, with an overload that receives a Consumer<String> for team logging, etc.

calcmogul, 2020-03-24:
No, Watchdog output should definitely be using reportWarning().

Starlight220, 2020-03-24:
That change is in #2413 . In the Tracer refactor, I am only setting the Tracer output to a lambda that defaults to DS.reportWarning().

calcmogul, 2020-03-24:
Ok, that sounds different than what you said originally.

No matter how you do this, it'll conflict with #2413 unless you base it on top of #2413.

Starlight220, 2020-03-24:
I didn't base it on #2413 so they won't be dependent, and because you said to keep them separate. Actually, it is much easier to do the refactor of #2413 after this (#2452), so perhaps I should do #2413 on top of #2452 so they can be merged together.

PR #2452 (2020-04-02): [wpilibj] Refactor Tracer functionality out of Watchdog class
resolves #2379 .
Needs C++ version.
closed 2020-04-02 by PeterJohnson