wpilibsuite/allwpilib #2383: Consistency with controllers

JohnnyWobble, 2020-02-21:
Within the `XboxController` class, the `getX(hand)` and `getY(hand)` are public methods. But in the `Joystick` class it inherits `getX` and `getY` from `GenericHID` where `getX` and `getY` are `public final`. 
I am trying to subclass `Joystick` to add a deadband and an x^2 curve to the output, and I am finding that having the methods being `final` is not helpful. My plan was to have an interface for deadbands and curves, but this is made more difficult by having some of the methods `final` and some not.
I think either make the `XboxController` methods `final` as well, or make all of them not `final`. Maybe even implements deadbands and curves in the WPIlib controllers as well.

Starlight220, 2020-04-20:
> Maybe even implements deadbands and curves in the WPIlib controllers as well.

I think this already exists. Check `GenericHID`.

calcmogul, 2020-08-26:
It doesn't, but it should. We could move the deadband logic from the drive classes to GenericHID like #1691 is doing, but in a separate PR. This makes sense to do because one would want the deadband applied to any accesses of an axis. The same is not necessarily true for squaring.

PR #3529 (2021-08-29): Move RobotDriveBase::ApplyDeadband() to MathUtil
It's a useful function outside of the drive classes.

Closes #2383.
closed 2021-08-29 by PeterJohnson