wpilibsuite/allwpilib #2384: std++fs support

sjcbulldog, 2020-02-21:
This is probably a moot point since this year is almost done and next year may include a newer GCC compiler with real C++17 support, but any plans to get the stdc++fs library compiled for G++ so that we can use the <experimental/filesystem> support on the roborio? This would be beneficial for us even after the season for post season projects.

PeterJohnson, 2020-02-22:
There are no plans to do that.  Have you looked at the wpiutil wpi/Filesystem.h header?  It likely has most of what you might want.

sjcbulldog, 2020-02-22:
It can work.  We have some code that is dependent on WPILIB based code.  We have other code that is independent of WPILIB and that is where the code is where we want to use the filesystem related calls.  We will just #ifdef it wincows vs linux.

closed 2020-09-13 by AustinShalit