wpilibsuite/allwpilib #2385: Field2d not importing correctly

retrodaredevil, 2020-02-21:
After updating to wpilib version 2020.3.2 I am unable to import `Field2d` from the `edu.wpi.first.wpilibj.simulation` package. However, for some reason if I created a kotlin file, I can import `Field2d` by just having `import Field2d` at the top. 

I don't know why `Field2d` isn't in its correct package, but it doesn't seem like the `edu.wpi.first.wpilibj.simulation` package even exists.

ThadHouse, 2020-02-22:
It looks like the package declaration was not added to the file, which means it looks like a default packaged class, but the classloader won't find it. 

There are no other classes in the simulation package, which is why it doesn't show up.

PeterJohnson, 2020-02-22:
Fixed in #2386.

closed 2020-02-22 by PeterJohnson