wpilibsuite/allwpilib #2388: Calling a command's cancel() method in its initialize() method doesn't cancel the command

chenyx512, 2020-02-24:
In the new command-based rewrite (Java at least), the CommandScheduler's cancel() method requires the command to be in m_scheduledCommands.
However, the initCommand() method calls command.initialize() before putting in into m_scheduledCommand(). 
Therefore, when I want to perform some precheck in a command's initialize() method and cancle the command using this.cancle() if the check fails, the command will not be canceled.

Starlight220, 2020-03-16:
This is something that should not be changed in a season. 
Idea for a workaround: have a boolean variable that is set in initialize and cancels the command first thing in execute.

Starlight220, 2020-03-22:
I will start Java fix soon. Someone should check if this bug exists in C++.

LincolnSand, 2020-03-29:
I have not observed this issue in C++

Starlight220, 2020-03-29:
> I have not observed this issue in C++

So there is no reason to fix C++ version. As I see this, it is simply a difference in the order in the lines of code.

PR #2440 (2020-05-21): [command] Make cancel safe to call from initialize
fixes #2388 and updates docs.
commit 92380485 (2020-05-21): [command] Make cancel safe to call from initialize (#2440)

Fixes #2388.
closed 2020-05-21 by PeterJohnson