wpilibsuite/allwpilib #2389: Make PerpetualCommand reinitialize the if the Giving command is CommandGroup

nimrod46, 2020-02-24:
Using `PerpetualCommand` with a given `CommandGroup` would not make the command group reschedule his commands (did not test it but its looks that way from the design) which brakes the idea of PerpetualCommand.

`whileActiveContinuous` will make the same result and solve the problem but what if the driver would not want to hole the button? I will have to right a new command to make the same logic.
 or
If I need this command to run in auto mode and I used the same command that I gave to `whileActiveContinuous` the command would not work the same way which makes me wonder, do we really need to confuse beginners with this approche?

Not saying we should rush to remove `whileActiveContinuous` in the middle of the season but its worth discussing it. Fixing `PerpetualCommand` (or crating a completely new command for that so we won't effect anyone using it as it is) is mandatory In my opinion (unless there is a simpler way doing it which I meassed?).

Related links:
https://github.com/wpilibsuite/allwpilib/issues/2302

https://www.chiefdelphi.com/t/running-command-constantly/378756/14

Starlight220, 2020-03-16:
In #2302 and the CD link you added I answered this issue, another solution/workaround that is more specific to this situation (whileActiveContinuous) is using the Trigger class.

Starlight220, 2020-04-12:
Before work starts on this, see the discussion in https://github.com/wpilibsuite/allwpilib/pull/2451#issuecomment-612659834. The fix for this should follow the decision made there.

closed 2022-04-21 by nimrod46