wpilibsuite/allwpilib #2390: Standardize on a gyro rotation positive direction and stick with it

jasondaming, 2020-02-25:
Currently the odometry stuff uses counterclockwise positive and arcade drive uses clockwise positive.  It is confusing to have two different standards in the same toolset.  I think there can be arguments made to using either one.  The worst decision IMO however is to not standardize!

calcmogul, 2020-02-25:
Well, joysticks use NED axes convention and mathematics uses NWU axes convention. We've been using the latter for all the kinematics and trajectory stuff.

Arcade drive and others use NED axes to match joysticks since they are most commonly used together. If we move to NWU, teams will need some well-placed negative signs (not that they don't already throw negative signs everywhere and hope it sticks).

retrodaredevil, 2020-03-21:
I think we need a new Gyro interface to work with. Right now `getAngle()` returns degrees, which is fine, but now we also have the `Rotation2d` class we should be taking advantage of. Maybe if we were to create a new Gyro class the new class could have a `getHeading()` method which returns a `Rotation2d`?

Maybe this new class could go in the frc2 or wpilibj2 package and still be called `Gyro`. 

I agree that the joystick stuff should stay the way it is, but IMO other stuff not joystick related should use NWU. And maybe for the stuff that uses NED methods (including joystick stuff) could be added to add explicit support for NWU values. I'm not really sure how that would work, though.

calcmogul, 2020-08-26:
Fixed by #2555.

closed 2020-08-26 by calcmogul