wpilibsuite/allwpilib #2393: SplineHelper.java (maybe c++ too) -- getCubicSplinesFromControlVectors

jsimpso81, 2020-02-27:
There appears to be an issue in the method  getCubicSplinesFromControlVectors of class SplineHelper.

In this code:

      if (newWaypts.length > 4) {
        for (int i = 1; i <= newWaypts.length - 4; i++) {
          dx[i] = 3 * (newWaypts[i + 1].getX() - newWaypts[i - 1].getX());
          dy[i] = 3 * (newWaypts[i + 1].getY() - newWaypts[i - 1].getY());
        }
      }

Either the variable newWaypts was intended to be waypoints or the index into the newWaypts array needs to have 1 added to all four instances in this loop (to account for the dx and dy being internal waypoints and newWaypts including the start and end.

calcmogul, 2020-02-27:
What's the expected behavior, and what did it do instead?

jsimpso81, 2020-02-28:
I responded via email, but essentially, I have a test program that returned the following results.  
[image]

It should have looked like this.

[image]

Note that this is NOT in JAVA, but this implements the code.... Suggest implementing a java test program to see if the problem exists in the java. (I assumed it was my code at first.... Once I fixed my code it pointed to there being a potential issue with the java too..)

Pseudo code for the test program.
- enter initial position x, y, heading  (2,0,PI/2) POSE
  - enter final position x, y, heading (2,0,PI/2) POSE
  - enter waypoints (starting at cos( PI/8 ) * 2,  sin( PI/8 ) * 2
         (waypoints all the way around every PI/8, not including initial and final)
  - Call splineHelp - GetCubicControlVectorsFromWayPoints
  - Call splineHelp - GetCubicSplinesFromControLVectors
  - Call TrajectoryGenerate - splinePointsFromSplines (to plot)
  - plot the results.

Hope this helps.

PR #2572 (2020-07-06): [wpilib] Fix SplineHelper cubic spline bug
Closes #2393
closed 2020-07-06 by PeterJohnson