wpilibsuite/allwpilib #2395: Simulator GUI won't render/startup properly

28andrew, 2020-02-28:
Recently I have not been able to get the Simulator GUI to be functional. It loads up and shows up in my task bar. However, when I click on it, no window shows up. I have updated to 2020.3.2 and still have the same exact issue.

The gradle simulator task log:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS D:\Programming\FRC\2020InfiniteRecharge> & 'C:\Users\Public\wpilib\2020\vscode\data\extensions\vscjava.vscode-java-debug-0.24.0\scripts\launcher.bat' 'C:\Users\Public\wpilib\2020\jdk\bin\java' '-agentlib:jdwp=transport=dt_socket,server=n,suspend=y,address=localhost:56917' '-Djava.library.path=D:\Programming\FRC\2020InfiniteRecharge\build\tmp\jniExtractDir' '-Dfile.encoding=UTF-8' '@C:\Users\ANDREW~1\AppData\Local\Temp\cp_7q3c77qmo4e1yuluh1zv42jdl.argfile' 'frc.robot.Main'
'chcp.com' is not recognized as an internal or external command,
operable program or batch file.
HAL Extensions: Attempting to load: halsim_gui
Simulator GUI Initializing.
windowScale = 1.000000e+00Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
```

What the window looks like in Alt-Tab:
[image]

Daltz333, 2020-02-28:
If your team uses C++, an additional dependency is required as per https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/robot-simulation/introduction.html#additional-c-dependency

[image]

PeterJohnson, 2020-02-28:
This looks like a graphics card / driver issue with OpenGL rendering.  What kind of graphics card?  Are the drivers up to date?  etc.

PeterJohnson, 2020-02-29:
Does it work on a different computer?

28andrew, 2020-02-29:
Hmm, it works perfectly fine on my other computer. Is there a way for me to reinstall WPILib?

PeterJohnson, 2020-02-29:
I very much doubt reinstalling WPILib will help.  This is a computer/OS/graphics card issue.  The GUI uses OpenGL to render and something is broken on this particular machine with that rendering pipeline.

28andrew, 2020-03-02:
Oddly I made a new project from a template of TimedRobot and the GUI worked just fine. So I tried my robot code again and it didn't work. Then I commented out all subsystem instantiation in RobotContainer and the problem persists. Very bizarre.

28andrew, 2020-03-15:
I've found the issue! It turns out the imgui.ini that was present was causing the issue. I don't know if this was because of just a bad value, updating WPILib versions, unplugging a joystick, or unplugging a camera, or something else. For posterity:

Problematic `imgui.ini`:
```
[Window][Debug##Default]
Pos=60,60
Size=400,400
Collapsed=0

[Window][Robot State]
Pos=5,20
Size=92,99
Collapsed=0

[Window][Analog Inputs]
Pos=640,20
Size=204,65
Collapsed=0

[Window][FMS]
Pos=5,540
Size=235,169
Collapsed=0

[Window][Joysticks]
Pos=251,465
Size=796,73
Collapsed=0

[Window][DIO]
Pos=470,20
Size=104,48
Collapsed=0

[Window][Encoders]
Pos=640,215
Size=85,48
Collapsed=0

[Window][PWM Outputs]
Pos=910,20
Size=104,48
Collapsed=0

[Window][Relays]
Pos=180,20
Size=73,48
Collapsed=0

[Window][Other Devices]
Pos=1025,20
Size=250,695
Collapsed=0

[Window][Solenoids]
Pos=290,20
Size=92,48
Collapsed=0

[Window][Timing]
Pos=5,150
Size=135,104
Collapsed=0

[Window][System Joysticks]
Pos=40,388
Size=192,150
Collapsed=0

[SimWindow][GLOBAL]
width=1280
height=720
maximized=0
xpos=-32000
ypos=-32000
userScale=2
style=0
disableOutputsOnDS=1
[SimWindow][Addressable LEDs]
visible=0
enabled=1

[SimWindow][Analog Inputs]
visible=1
enabled=1

[SimWindow][FMS]
visible=1
enabled=1

[SimWindow][System Joysticks]
visible=1
enabled=1

[SimWindow][Joysticks]
visible=1
enabled=1

[SimWindow][DIO]
visible=1
enabled=1

[SimWindow][Encoders]
visible=1
enabled=1

[SimWindow][2D Field Settings]
visible=0
enabled=1

[SimWindow][2D Field View]
visible=0
enabled=1

[SimWindow][PDP]
visible=0
enabled=1

[SimWindow][PWM Outputs]
visible=1
enabled=1

[SimWindow][Relays]
visible=1
enabled=1

[SimWindow][RoboRIO]
visible=0
enabled=1

[SimWindow][Other Devices]
visible=1
enabled=1

[SimWindow][Solenoids]
visible=1
enabled=1

[SimWindow][Timing]
visible=1
enabled=1

[DriverStation][Main]
disable=0

[Field2D][Field]
image=
top=0
left=0
bottom=-1
right=-1
width=15.980000
height=8.210000

[Field2D][Robot]
image=
width=0.685800
length=0.820400

[Device][Talon SRX[58]]
open=1

[Device][Talon SRX[20]]
open=0

[Device][Talon SRX[12]]
open=1

[Device][Talon SRX[60]]
open=1

[Device][Talon SRX[44]]
open=1

[Device][navX-Sensor[0]]
open=0

[Device][REV Color Sensor V3[0,82]]
open=0
````

Default `imgui.ini`, created after deleting problematic one:
```
[Window][Debug##Default]
Pos=60,60
Size=400,400
Collapsed=0

[Window][Robot State]
Pos=5,20
Size=92,99
Collapsed=0

[Window][Analog Inputs]
Pos=640,20
Size=204,65
Collapsed=0

[Window][FMS]
Pos=5,540
Size=235,169
Collapsed=0

[Window][System Joysticks]
Pos=5,385
Size=192,150
Collapsed=0

[Window][Joysticks]
Pos=250,465
Size=796,73
Collapsed=0

[Window][DIO]
Pos=470,20
Size=162,77
Collapsed=0

[Window][Encoders]
Pos=640,215
Size=85,48
Collapsed=0

[Window][PWM Outputs]
Pos=910,20
Size=104,48
Collapsed=0

[Window][Relays]
Pos=180,20
Size=73,48
Collapsed=0

[Window][Other Devices]
Pos=1025,20
Size=250,695
Collapsed=0

[Window][Solenoids]
Pos=290,20
Size=92,48
Collapsed=0

[Window][Timing]
Pos=5,150
Size=135,104
Collapsed=0

[SimWindow][GLOBAL]
width=1280
height=720
maximized=0
xpos=-1
ypos=-1
userScale=2
style=0
disableOutputsOnDS=1
[SimWindow][Addressable LEDs]
visible=0
enabled=1

[SimWindow][Analog Inputs]
visible=1
enabled=1

[SimWindow][FMS]
visible=1
enabled=1

[SimWindow][System Joysticks]
visible=1
enabled=1

[SimWindow][Joysticks]
visible=1
enabled=1

[SimWindow][DIO]
visible=1
enabled=1

[SimWindow][Encoders]
visible=1
enabled=1

[SimWindow][2D Field Settings]
visible=0
enabled=1

[SimWindow][2D Field View]
visible=0
enabled=1

[SimWindow][PDP]
visible=0
enabled=1

[SimWindow][PWM Outputs]
visible=1
enabled=1

[SimWindow][Relays]
visible=1
enabled=1

[SimWindow][RoboRIO]
visible=0
enabled=1

[SimWindow][Other Devices]
visible=1
enabled=1

[SimWindow][Solenoids]
visible=1
enabled=1

[SimWindow][Timing]
visible=1
enabled=1

[DriverStation][Main]
disable=0

[DIO][0]
name=

[DIO][1]
name=

[Field2D][Field]
image=
top=0
left=0
bottom=-1
right=-1
width=15.980000
height=8.210000

[Field2D][Robot]
image=
width=0.685800
length=0.820400

[Device][Talon SRX[58]]
open=0

[Device][Talon SRX[20]]
open=0

[Device][Talon SRX[12]]
open=0

[Device][Talon SRX[60]]
open=0

[Device][Talon SRX[44]]
open=0

[Device][REV Color Sensor V3[0,82]]
open=0
```

Diff: https://gist.github.com/28andrew/99656b4132f4ac768badb60435b4383d/revisions

*I can reproduce this bug by copying the problematic `imgui.ini` to a new project and trying to run the simulator GUI in it.*

auscompgeek, 2020-03-15:
Looks like the SimWindow global xpos and ypos are well off-screen. I wonder how that happened.

AustinShalit, 2020-09-13:
Closed by #2660. Windows now uses DirectX instead of OpenGL.

closed 2020-09-13 by AustinShalit