wpilibsuite/allwpilib #2396: wpiutil c++ compile errors

Droid22, 2020-02-29:
cmake version 3.5.1
repo version 2020.3.2 (last tag).
build options:
WITHOUT_JAVA on
WITHOUT_CSCORE on
BUILD_SHARED_LIBS on
WITHOUT_ALLWPILIB on
USE_EXTERNAL_HAL off

In the cmake file for wpiutil there was an error: "target_compile_features specified unknown feature "cxx_std_17" for target "wpiutil"

I tried changing it to cxx_std_14 and also 11. Neither worked. Using this [article](https://stackoverflow.com/questions/42834844/how-to-get-cmake-to-pass-either-std-c14-c1y-or-c17-c1z-based-on-gcc-vers), changed code from 
`target_compile_features(wpiutil PUBLIC cxx_std_17)` to 
```
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
```
Would not generate cmake code for 17, generated code for 14 and 11. 
For both 14 and 11 I got compile errors of: 
use of c++ 11 features 
#error this file requires file and library support for c++ 11.

ThadHouse, 2020-02-29:
What system are you trying to compile on? Does it support c++ 17? That is a requirement, older versions of the standard will not work.

PeterJohnson, 2020-02-29:
cmake 3.5 is ancient.  You need to use 3.8 or newer.