wpilibsuite/allwpilib #2398: Add examples that utilize SimulationInit/SimulationPeriodic

virtuald, 2020-03-01:
Maybe update templates too?

Starlight220, 2020-04-19:
What exactly should the examples include?

virtuald, 2020-04-19:
Ideally, a robot that is usable in simulation. See https://github.com/robotpy/examples/tree/master/physics/src .. RobotPy automatically hooks simulationInit/simulationPeriodic, but it utilizes the various simulation HAL objects to make the simulation actually useful. 

For example, see https://github.com/robotpy/examples/blob/418f4770550d6d8d56a9a54a672eb89837c252cd/physics/src/robot.py#L51 ... where the movement of a motor activates a limit switch, which allows the robot code to respond to that input and stop the motor. It's pretty trivial, but hopefully it helps teams understand the potential and then they can do more with this.

Starlight220, 2020-08-03:
When this is done, the example should also include using the SimDevice API. The problem is that the SimDevice isn't developed enough or not documented, therefore no one knows how to use it. 
There most documentation I've seen about the SimDevice is the vendor simulation documentation from wpilibsuite/frc-docs#350, and it covers the vendor/device end, not the team code end. 
If the SimDevice API is fully developed for Java, I can experiment with it a bit and add an example. 
@PeterJohnson - is the Java SimDevice API and all relevant classes ready for usage?

virtuald, 2020-08-03:
The SimDevice API should be fully usable from Java AFAIK. 

Here's a python example of retrieving the gyro value from a NavX sensor via SimDevice: https://github.com/robotpy/examples/blob/418f4770550d6d8d56a9a54a672eb89837c252cd/navx-rotate-to-angle/physics.py#L22

Starlight220, 2023-01-15:
How is this different than #2740?

Starlight220, 2023-09-28:
We have examples that showcase simulationInit/Periodic, and various other issues for cleanup and enhancement of examples. Petition to close this?

closed 2023-12-22 by Starlight220