wpilibsuite/allwpilib #2400: Rotation Only Trajectories Not Supported

AlexanderPeters, 2020-03-02:
It is currently impossible to make a robot spin in place or turn to an angle with a RamseteCommand. This is because the starting and ending point of the desired trajectory have the same x and y coordinates, only being rotated in between. Thus a spline can not be fit two the two points. I am wondering if this case can be handled. 

e.x.

`       var startingPose = new Pose2d(new Translation2d(0, 0), new Rotation2d(0));
        var endingPose = new Pose2d(new Translation2d(0, 0), new Rotation2d(angle));
        var interiorWaypoints = new ArrayList<Translation2d>();

        TrajectoryConfig config = new 
                TrajectoryConfig(Constants.DriveConstants.trajectoryMaxVelocity,
                        Constants.DriveConstants.trajectoryMaxAcceleration);

        Trajectory trajectory = TrajectoryGenerator.generateTrajectory(startingPose, interiorWaypoints, 
                endingPose, config);`

calcmogul, 2020-03-06:
We can special-case it with a trapezoid profile on heading. The problem is none of the trajectory configuration takes rotational dynamics into account (max angular velocity and max angular acceleration).

We sort of have a hacky solution in DifferentialDriveVoltageConstraint by assuming linear and angular dynamics are decoupled. However, for accurate constraints, it isn't safe to assume the linear and angular Kv and Ka from which we can calculate max dynamics would be the same. In my team's case, they are different by a factor of two.

https://github.com/frc3512/Robot-2020/blob/master/src/main/include/controllers/DrivetrainController.hpp#L219-L224

We have a custom constraint that enforces the dynamics of a 2x2 MIMO drivetrain model. It works well, but still doesn't fix the numerical conditioning issues caused by infinite curvature. We'll need to revisit this for 2021.

AlexanderPeters, 2020-03-06:
Thanks, I'll take a looks at that. It would be nice to see the robot characterization took expanded to generate angular parameters in addition to linear.

prateekma, 2020-11-10:
Unfortunately, with the way that our trajectories work, this is not possible. Because angular velocity is calculated by feedback / feedforward controllers by multiplying the linear velocity of a particular trajectory state by its curvature, the calculated angular velocity will always be zero when the linear velocity is zero.

The best alternative to this is to just use the `TrapezoidProfile` class.

closed 2020-11-11 by AustinShalit

pietroglyph, 2020-12-08:
> It would be nice to see the robot characterization took expanded to generate angular parameters in addition to linear.

I know I'm a bit late to the party here, but I think it's worth mentioning that the characterization tool *does* support generating angular parameters if you check the "Angular Mode" checkbox in the data logger tool. This will make the robot spin in place, and you'll end up with angular K_s/K_v/K_a.