wpilibsuite/allwpilib #2401: Notifier alarmCallback may fail to schedule the next alarm in some circumstances

chauser, 2020-03-03:
We sporadically observe our code's main thread hanging -- no more dashboard updates, no flashing LEDs, no console messages from the main thread -- for tens of minutes. If you give it long enough sometimes it starts working again.
I have attached gdb to the process and find that when it is stuck the call stack looks like:
[image]
`TimedRobot::StartCompetition` calls `HAL_WaitForNotifierAlarm` to pause before starting the next loop of the main robot code, and apparently it doesn't wake up as expected after 10ms (in our case).

In the Notifier `alarmCallback` code I found this:
```
  // process all notifiers                                                                                                                                                                                           
  notifierHandles->ForEach([&](HAL_NotifierHandle handle, Notifier* notifier) {
    if (notifier->triggerTime == UINT64_MAX) return;
    if (currentTime == 0) currentTime = HAL_GetFPGATime(&status);
    std::unique_lock lock(notifier->mutex);
    if (notifier->triggerTime < currentTime) {
      notifier->triggerTime = UINT64_MAX;
      notifier->triggeredTime = currentTime;
      lock.unlock();
      notifier->cond.notify_all();
    } else if (notifier->triggerTime < closestTrigger) {
      closestTrigger = notifier->triggerTime;
    }
  });
```
It looks to be a problem that upon finding a notifier in the handles collection that does not require further processing the code executes a `return` instead of a `continue`. It thus ignores any later notifiers that may need an alarm to be scheduled. 
Of course this can only happen in there are multiple notifiers being used. I don't find any other notifiers in the wpilibc code and our code doesn't use any. But there could be one or more in the non-public code associated with the DriveStation interaction. And I do observe that the hang is likely to occur after the Drive Station has been disconnected for quite a while (10s of minutes).

PeterJohnson, 2020-03-03:
The HAL Notifier is only used in WPILib (or potentially in vendor libraries) and not in any lower level code (e.g. the NI code that talks to the DS).

chauser, 2020-03-03:
Thanks, Peter. Wish I could figure out why it's not waking up!

PeterJohnson, 2020-03-03:
Actually it’s not a big.  Return is the correct thing.  This is a functor not a normal loop.

chauser, 2020-03-03:
I see. Sorry I missed that. Guess I'll keep looking for how it can hang at the place that it did.

PeterJohnson, 2020-03-03:
Well, if you just hit Ctrl-C, that's the most probable place to stop.  Does single stepping exit the function?  A link to your code would be helpful.

chauser, 2020-03-03:
Hi Peter, it's the same code and repo (sciborgs) I sent you before for the
rio crashing/loss of comms problem. I thought of the "most likely place to
stop" idea as well -- and that's true if the code is running normally. But
since the main thread is clearly hung somewhere (none of the things it does
are happening and cpu utilization is also way down) that is strongly
suggestive that it is truly hung in the spot where gdb caught it. I've
observed this occurring now several times and been able to attach gdb to a
few of them and each time it has been the same.

So anyway, I will close this issue -- unless you already did -- as it is
clear that the particular code it talks about is in fact ok.

Next time I get my hands on the boardbot I will set it going again and
carefully watch the power-on time at which it occurs. Given the impressions
about "it's a long time before it happens" and "then it recovers after a
long time" I have a sneaking suspicion it is related to the rollover of the
fpga clock around 71 minutes. I'm concerned because the Notifier code
treats the fpga timestamp as if it were really 64 bits (as the
HAL_GetFPGATime declaration states), but the documentation in the Timer
class says that the timestamps roll over at 71 minutes. The latter would
not need to be true if the timestamps were really 64 bits.

And if this were indeed the way the problem manifests, I could stop
worrying about it because we can certainly arrange for the robot to be up
less than 71 minutes in competition.

Thanks!

closed 2020-03-03 by chauser

chauser, 2020-08-09:
I should explain what was happening in case anyone else encounters it. Suppose you have a periodic Notifier with a delay time t, but the code that is run each time takes more than t. Then, since the Notifier code is designed to try to catch up in this situation, the next wakeup time is scheduled t after the first wakeup, but that time has already past. As time passes, the wakeup times get farther and farther in the past, until they are more than 1/2 the clock rollover period behind the current time. At that point, the scheduled wakeup time now appears to be 1/2 the rollover period _in the future_ and everything stops until then. This is about 36 minutes. If you let the system stay up that long it will resume. 

How long it takes for the wakeups to stop depends on by how much the average time for the notifier's code exceeds the declared time. But the time to wakeup again is reliably 1/2 the clock rollover period.