wpilibsuite/allwpilib #2402: Disconnected PCM with DoubleSolenoid crashes robot code

Modelmat, 2020-03-03:
This was the error received through DriverStation.
[driverstation-logs.zip](https://github.com/wpilibsuite/allwpilib/files/4280884/driverstation-logs.zip)

```
05:14:56.160
   	at __libc_start_main  	at int frc::StartRobot<Robot>()
  	at frc::TimedRobot::StartCompetition()
  	at frc::IterativeRobotBase::LoopFunc()
  	at frc::LiveWindow::UpdateValues()
  	at frc::LiveWindow::UpdateValuesUnsafe()
  	at frc::SendableRegistry::ForeachLiveWindow(int, wpi::function_ref<void (frc::SendableRegistry::CallbackData&)>) const
  	at   	at frc::SendableBuilderImpl::UpdateTable()
  	at   	at   	at frc::DoubleSolenoid::Get() const
  Error at Get [DoubleSolenoid.cpp:125]: CTRE CAN Receive Timeout Warning
```
This was a setup with the PCM and Pneumatics disconnected.
I am going to test but I believe this is because the PCM is disconnected, we have had pneumatics disconnected previously but it did not crash robot code.

If I'm reading the source correctly this should not be happening - `StatusIsFatal()` should be returning true. https://github.com/wpilibsuite/allwpilib/blob/9bcff37b93b2ca22b3dfbf9ac198d46f3ff38895/wpilibc/src/main/native/cpp/DoubleSolenoid.cpp#L119

The code involved was:
```cpp
frc::DoubleSolenoid m_pneumatic{kPneumaticPorts[0], kPneumaticPorts[1]};
void IntakeSubsystem::DeployIntake() {
  m_pneumatic.Set(frc::DoubleSolenoid::kForward);
}
void IntakeSubsystem::StowIntake() {
  StopIntake();
  m_pneumatic.Set(frc::DoubleSolenoid::kReverse);
}
void IntakeSubsystem::ToggleDeployed() {
  switch (m_pneumatic.Get()) {
  case frc::DoubleSolenoid::kForward:
    StowIntake();
    break;
  case frc::DoubleSolenoid::kReverse:
    DeployIntake();
    break;
  case frc::DoubleSolenoid::kOff:
    break;
  }
}
```

Googling this error gives results akin to `your PCM is disconnected, connect it`.
I believe it would be better here to *not crash* the robot code and instead just no-op, as encoders etc. do they fail to connect.

auscompgeek, 2020-03-03:
Do you have any other CAN devices in your code?

Kevin-OConnor, 2020-03-03:
An "Error" is a method of reporting to the DriverStation, it does not cause code to quit.

Unfortunately, if you would like stack traces like this for your crashes in FRC you need to be using Java. C++ just bombs out with no indication as to why.

I will leave this open and try to test this particular code path soon, but I doubt this is the source of the crash

Modelmat, 2020-03-03:
I don't necessarily want stack traces, I would just like for the code not to crash.

We do have motor controllers over CAN connected, and could see them all and control them once we removed the pneumatic code.

I will confirm later today but the code was definitely crashing a and stopped doing so after removing the pneumatics code

Modelmat, 2020-03-04:
Well, it appears that this was my bad.
It dosnt' seem to be crashg the code at all.

closed 2020-03-04 by Modelmat