wpilibsuite/allwpilib #2408: Feedforward/PID integration

KevinDowning, 2020-03-12:
Have a set of classes that run the tests for finding feedforward and PID values which are sent to a csv file so that a separate program doesn't have to be used.

Starlight220, 2020-03-15:
In other words: you want a PID tuning tool similar to the Robot Characterization Tool?
Because it shouldn't be in the library...

calcmogul, 2020-03-15:
As @Starlight220 said, having that functionality is a violation of separation of concerns. If we added it to wpilib, you'd need a separate copy of your code to run the characterization. frc-characterization already does that automatically, so the tool should be preferred.

For PID tuning though, we'll have a way to produce PD gains from feedforward gains in 2021 the same way frc-characterization does it. It'll come as part of the state-space library. The prototype API looks like this:

```cpp
#include <frc/controller/LinearQuadraticRegulator.h>
#include <frc/system/plant/LinearSystem.h>
#include <frc/system/plant/LinearSystemId.h>

frc::LinearSystem<2, 1, 1> plant = frc::IdentifyPositionSystem(kV, kA);
frc::LinearQuadraticRegulator<2, 1> lqr{m_plant, {0.3, 3.0}, {12.0}, 5_ms};

double Kp = lqr.K(0, 0);
double Kd = lqr.K(0, 1);
```
The tradeoff values passed into the LQR constructor are the same as what's in the frc-characterization GUI.

Starlight220, 2020-03-15:
Is there a reason this issue is still open? If not, should be closed.