wpilibsuite/allwpilib #2409: LiveWindow,.updateValues takes too long in Simulation

blu28, 2020-03-13:
When running the desktop simulation I often get Watch dof not fed and loop overun messages. When the time epochs are dumped out it is invariably LiveWindow.updateValues that is the culprit. This even happens on small robot codes that have no calls to LiveWindow or Shuffleboard logging of any kind.

PeterJohnson, 2020-03-13:
Is this a Java or C++ program?  What kind of computer specs (e.g. CPU)?

sciencewhiz, 2020-03-13:
Also, which devices are you using in the simple programs?

blu28, 2020-03-13:
Java. 
Intel i7-8700K, 3.7GHz, 6 cores, 32GB memory.

Here is my RobotContainer class:

```
public class RobotContainer {
    private final PWMVictorSPX m_leftmotor = new PWMVictorSPX(DriveConstants.kLeftMotorPort);
    private final PWMVictorSPX m_rightmotor = new PWMVictorSPX(DriveConstants.kRightMotorPort);
    private final DriveSubsystem m_drivesubsystem = new DriveSubsystem(m_leftmotor, m_rightmotor);
    private final GenericHID m_stick = new Joystick(OIConstants.kJoystickPort);
    private final Command m_autocommand;
    private final Command m_drivecommand;
    
    public RobotContainer() {
        m_autocommand = new AutoCommand(m_drivesubsystem);
        m_drivecommand = new DriveCommand(m_drivesubsystem, m_stick);
        m_drivesubsystem.setDefaultCommand(m_drivecommand);
    }
    public Command getAutonomousCommand() {
        return m_autocommand;
    }
}
```

I just tried running several simulations in a row, and there seems to be a pattern to it. There are a few overruns as the simulation starts up. But if I let it sit for a while in disabled mode they go away. There is always an overrun immediately at startup and then again within a second or two. This stops and if I just let the robot sit in disabled mode I don't see anymore. This would suggest that the overruns are associated with initialization or constructors. I am pretty sure I have seem the overruns later though, but I can't reproduce that at the moment.

carbotaniuman, 2020-08-02:
This is also very likely to be associated with Java GC pressure and JIT delays, so keep that in mind. (An empty robot project overruns frequently on the Rio).

Starlight220, 2023-09-26:
Are there any relevant actionable items around this issue?

blu28, 2023-09-26:
Perhaps suppress printing them when the simulation starts in disabled mode and only enable them after some number of iterations.

Starlight220, 2023-09-27:
That doesn't seem to relate directly from the issue here...

Starlight220, 2023-12-22:
Closing due to no action items.

closed 2023-12-22 by Starlight220