wpilibsuite/allwpilib #2411: Javadocs links to java 11 docs broken

PeterJohnson, 2020-03-14:
The generated javadocs links to Java objects (e.g. Object) are incorrect, as they point to "java" rather than "java.base".  E.g. https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/hal/sim/AccelerometerSim.html
The "Object" link points to https://docs.oracle.com/en/java/javase/11/docs/api/java/lang/Object.html?is-external=true
but should point to https://docs.oracle.com/en/java/javase/11/docs/api/java.base/java/lang/Object.html?is-external=true

May be due to this issue, but may also need configuration changes: https://bugs.openjdk.java.net/browse/JDK-8205593

Starlight220, 2020-03-16:
Should I work on this? 
If yes, any ideas for course of action?
Mine is simply checking that all links point to the correct place...

Starlight220, 2020-03-16:
This is a better-explained duplicate of #2308 .

AustinShalit, 2020-03-20:
Duplicate of #2308 

@Starlight220 To answer your question, it looks like this fix was never back ported to jdk11. It seems like the fix will need to be upgrading from java 11 to java12+

closed 2020-03-20 by AustinShalit