wpilibsuite/allwpilib #2427: Add 3D geometry classes

calcmogul, 2020-03-18:
These are useful for handling perspective n-point measurements from a vision pipeline within the WPILib geometry framework.

Translation3d is easy, but Rotation3d and Pose3d are less trivial. Making all three of these interoperate well with 2D may be hard.

Here's a mostly constexpr Translation3d: https://github.com/frc3512/Robot-2020/blob/master/thirdparty/include/frc/geometry/Translation3d.h

pietroglyph, 2020-03-18:
I have this on my list of wpilib items I’m planning to do, but if someone gets to it before me then so be it. Should be quick and easy to port Transform3d/Translation3d to Java. The other 3d classes are lower priority because their usecase isn’t as clear.

calcmogul, 2020-05-27:
We should consider looking at https://github.com/strasdat/Sophus/tree/master/sophus for an implementation of 3D rotation since our implementation of SE(2) is roughly based on it.

calcmogul, 2020-05-28:
https://github.com/strasdat/Sophus/blob/master/sophus/so3.hpp uses quaternions for SO(3).

m10653, 2022-02-28:
What is the progress on this issue. Is this something that would be be merged into WpiLib in the future given someone made a pull request implementing it?

calcmogul, 2022-02-28:
There's been no progress on this. Contributions are welcome.

m10653, 2022-02-28:
Perfect, thanks! I plan on it. Although I do plan on getting my feet wet
with a few of the good first issue labels first. Mainly wanted to make sure
I did not waste time on a feature that was rejected.

closed 2022-05-06 by PeterJohnson