wpilibsuite/allwpilib #2428: Failed to generate Trajectory when two pose close on a straight line

lingerer, 2020-03-18:
The first will fail and the second will work.

> 	const auto t = frc::TrajectoryGenerator::GenerateTrajectory(
		std::vector<Pose2d>{
		Pose2d(0_m, 0_m, Rotation2d(0_deg)),
		Pose2d(6_m, 0_m, Rotation2d(0_deg)),
		Pose2d(8_m, 0_m, Rotation2d(0_deg))},
		TrajectoryConfig(1_fps, 1_fps_sq));
	cout << t.States().size()<<","<<t.TotalTime()<<endl;
	const auto t1 = frc::TrajectoryGenerator::GenerateTrajectory(
		std::vector<Pose2d>{
		Pose2d(0_m, 0_m, Rotation2d(0_deg)),
		Pose2d(6_m, 0_m, Rotation2d(0_deg)),
		Pose2d(8.4_m, 0_m, Rotation2d(0_deg))},
		TrajectoryConfig(1_fps, 1_fps_sq));
	cout << t1.States().size() << "," << t1.TotalTime() << endl;

prateekma, 2020-04-02:
@PeterJohnson This can be closed now since #2429 was merged.

AustinShalit, 2020-04-02:
Closed by #2429

closed 2020-04-02 by AustinShalit