wpilibsuite/allwpilib #2438: Swerve and mecanum trajectory tracking should support arbitrary headings at endpoints

prateekma, 2020-03-22:
Instead of the heading of the last point on the trajectory, users should be able to specify custom headings for path tracking since this is achievable using a swerve or mecanum drive.

calcmogul, 2020-03-22:
The mecanum and swerve controller examples don't use the generated trajectory headings, since the heading is a free variable. They map the heading to a trapezoid profile instead, but the user can do whatever they want, really.

abhaybd, 2020-05-17:
@calcmogul The trajectory generation makes no distinction between heading of the robot and the direction of the velocity vector. Since these aren't the same for holonomic drivebases, it becomes a little unwieldy. For example, I don't think it's possible to generate a trajectory that goes straight forward while turning 90 degrees.

When using the SwerveControllerCommand, that makes it hard to follow holonomic paths.

I was able to mostly solve this using some really hacky fixes in my team's code, but I think it would be an easy fix to add a heading parameter to the SwerveControllerCommand, which would be the heading setpoint, instead of the heading of the last pose.

PR #2597 (2020-08-02): Add holonomic drive controller; make swerve and mecanum commands use new controller
Closes #2438
closed 2020-08-02 by PeterJohnson