wpilibsuite/allwpilib #2449: Move RGB->HSV Functionality Into Color Class

carlosgj, 2020-03-24:
Currently, the only wpilib HSV -> RGB converter is in AddressableLEDBuffer (setHSV). Therefore, the only way to initialize a Color object with HSV values is by going through an intermediate buffer:
```
AddressableLEDBuffer tempBuffer = new AddressableLEDBuffer(1);
tempBuffer.setHSV(0, 100, 255, 255);
Color myColor = tempBuffer.getLED(0);
```

HSV->RGB conversion should be handled in the Color class, either as a static function as in java.awt.Color.getHSBColor which returns a Color object, or as a method of a Color object.

Starlight220, 2020-03-24:
I would also add CYMK, as it would seem FIRST prefers it (it was used to define the Control Panel colors in the manual).

AustinShalit, 2020-03-24:
I would not recommend a converter for CYMK. CYMK is used for printing with ink and not for light. In order to write a converter, you would need to know not only the amount of each ink color other properties as well. If we are looking for a different color space to use, I would recommend the LAB color space because it is very linear and thus might perform better in changing lighting conditions.

PR #2461 (2020-04-06): [wpilib] Add Color.fromHSV() static helper
Closes #2449
closed 2020-04-06 by PeterJohnson