wpilibsuite/allwpilib #2459: add POVAxis

Starlight220, 2020-03-31:
One of the main FRC uses of POV is for opposing edges of the POV to be used for moving something in two directions (i.e. moving an elevator up and down). Maybe add a class for this? I can do the Java part myself, will need C++ .

calcmogul, 2020-03-31:
It's pretty simple to write code without a POVAxis class, and I think it expresses the intent more clearly.
```cpp
if (joystick.GetPOV() == 90) {
  motor.Set(1.0);
} else if (joystick.GetPOV() == 270) {
  motor.Set(-1.0);
} else {
  motor.Set(0.0);
}
```
Command-based also has POVButton for a similar purpose.

calcmogul, 2020-03-31:
One might also want to do
```cpp
if (joystick.GetPOV() == 0) {
  motor.Set(1.0);
} else if (joystick.GetPOV() == 180) {
  motor.Set(-1.0);
} else {
  motor.Set(0.0);
}
```. A POVAxis class doesn't support both cases well unless you had an enum to disambiguate them.

closed 2020-04-02 by Starlight220