wpilibsuite/allwpilib #2460: PortForwarder provides no indication of success or failure

virtuald, 2020-03-31:
Maybe add a callback w/events indicating new connections or errors or whatever?

PeterJohnson, 2020-03-31:
Do you really care to get events in code, or would printing something be sufficient?

virtuald, 2020-03-31:
Printing is sufficient.

Starlight220, 2020-04-01:
Maybe add a DS.reportInfo() [consistant with reportWarning/reportError]? Or the idea is for when a joystick is connected, etc. for a message to be printed?

auscompgeek, 2020-04-01:
@Starlight220 We can't use the HAL from wpiutil.

virtuald, 2020-04-01:
Yeah. And I'm actually using this not on a robot because it was simple and convenient. :)

virtuald, 2020-04-01:
Oh, and @Starlight220 this forwards network ports, not joystick ports. :)

Starlight220, 2020-04-02:
I was more talking about electric (I2C, DIO, PWM...) ports.. but thanks for the clear up.

PR #6697 (2024-06-05): [wpinet] Add indication of success/failure to PortForwarder
Resolves #2460
closed 2024-06-05 by PeterJohnson