wpilibsuite/allwpilib #2473: [wpilib] Add epoch print tests to Tracer

calcmogul, 2020-04-06:
Verifying the contents of epoch prints would have caught https://github.com/wpilibsuite/allwpilib/issues/2392.

Here's an example of verifying stdout contents with Google Test: https://github.com/wpilibsuite/allwpilib/blob/master/wpiutil/src/test/native/cpp/UnitsTest.cpp#L1307-L1310

Here's an example of verifying stdout contents with JUnit: https://stackoverflow.com/a/1119559/5484466

Starlight220, 2020-04-16:
I can do Java, how do I get output from DS.reportWarning()?

calcmogul, 2020-04-16:
In unit tests, ReportWarning() just prints to stderr, so you just need to capture console output.

Starlight220, 2020-05-06:
In Java, there is an overload that takes a lambda - can be used for testing. Dunno if C++ has this option as well.

Starlight220, 2020-09-14:
Would it be good enough to make sure that the output conforms to a certain regex, or is something more advanced required?

AustinShalit, 2020-09-14:
For Java, you might take a look to see if SystemRules was updated for JUnit 5: https://github.com/stefanbirkner/system-rules/issues/55