wpilibsuite/allwpilib #2482: Add a count-based DigitalGlitchFilter/Smoother

Starlight220, 2020-04-12:
The current DigitalGlitchFilter requires the measurement to stay constant for a period of time for the filtered reading to change. What I think should be added is a filter that takes the last `x` (digital) measurements and the filtered value is the more dominant value (true/false). I can implement this myself for wpilibj, will need help with wpilibc, if this should be added at all.

calcmogul, 2020-04-12:
We already have that. It's called MedianFilter.

Starlight220, 2020-04-12:
MedianFilter deals with numeric values, unlike both DigitalGlitchFilter and what I am suggesting (that deal with boolean values).

ThadHouse, 2020-04-12:
DigitalGlitchFilter is actually implemented in the FPGA, designed for debouncing sensors. Adding a count filter would need to be done a completely different way. It could be done separately, but not as part of DigitalGlitchFilter.

calcmogul, 2020-04-12:
MedianFilter works with binary values. True is one and false is zero.

Starlight220, 2020-04-13:
About MedianFilter, shouldn't the `get` and `update` operations be separate? 
Either way, as C++ implicitly converts between boolean and numeric values and Java doesn't, should a boolean overload/version be added to wpilibj?

calcmogul, 2020-04-13:
MedianFilter has just a Calculate() method like that on purpose; that's the intended usage. LinearFilter and PIDController are the same way.

Since conversions from int to boolean and back are tedious in Java, the two options I see are create a MedianFilterBoolean class, or make MedianFilter a generic type. I'd prefer the latter, but it's proving tricky to implement in CircularBuffer (MedianFilter uses CircularBuffer internally).

Starlight220, 2020-04-13:
My team implemented this (the boolean smoother) in the season with an array and a rotating index. We separated the `get()`/`update()` functions, and counted how many 'true' values we have in the array.

Starlight220, 2020-04-16:
Should a new BooleanMedianFilter class wrap an internal MedianFilter object and convert booleans to numbers and the other way around, or should it be based on a boolean array with a rotating index?

PR #3590 (2021-09-20): [wpilib] Add Debouncer
Supersedes #2358 with updates and cleanups.

Closes #2482 and closes #2487 because we shouldn't support both
time-based and count-based debouncing approaches.
commit 1ca383b2 (2021-09-20): Add Debouncer (#3590)

Supersedes #2358 with updates and cleanups.

Closes #2482 and closes #2487 because we shouldn't support both
time-based and count-based debouncing approaches.
closed 2021-09-20 by PeterJohnson