wpilibsuite/allwpilib #2483: [command-based] Move Subsystem classes to an inner package

Starlight220, 2020-04-13:
Move the base subsystem classes (Subsystem, SubsystemBase, PIDSubsystem, ProfiledPIDSubsystem) to an inner package (`command.subsystem` instead of `command`). This will reduce clutter in the `commands` package, etc.

PeterJohnson, 2020-04-13:
While this could have been done during the initial design, now that the library is in use by teams, it’s really too late to do this (unless we do it as part of a major breaking vendor library revision).  Java is not able to easily do shims for backwards compatibility when moving things into different packages, which is needed to not have this be a massively breaking change.

closed 2020-04-13 by PeterJohnson

Starlight220, 2020-04-13:
Maybe keep this in mind for the next major change in the library?