wpilibsuite/allwpilib #2484: [wpilib] Unclutter base packages

Starlight220, 2020-04-13:
Split more classes into sub-packages of wpilibj/wpilibc.

A few ideas : 
- motorcontrol
- sensors
- hid

etc.

calcmogul, 2020-04-13:
That would be nice, but it would break _everyone_ and require us to make shims for almost every class in the library.

Starlight220, 2020-04-13:
Many of the classes in the sub-packages were once in the root library. Can't this be done in stages, etc?

PeterJohnson, 2020-04-13:
We’ve done a few moves when the changes are breaking for other reasons.  Eg old vs new PIDController.  A lot of classes haven’t changed in nearly 10 years, and doing this wholesale isn’t practical in Java.  This is a case where backwards compatibility is more important than this type of cleanup.

Starlight220, 2020-04-13:
So perhaps have a guideline saying that new classes should (probably) not be put in the root package?

calcmogul, 2020-04-13:
We've already been doing that since 2016. A lot of the classes that fit into the categories you listed have existed since 2008.

Starlight220, 2020-04-13:
So should (for example) PR #2418 put the new QuadratureEncoder class in a new subpackage (`encoders` or `sensors`)?

Starlight220, 2020-04-13:
> We've already been doing that since 2016.

Perhaps add a warning to the pipeline?

Daltz333, 2020-04-13:
For a bit more direct communication, we do have a gitter. https://gitter.im/wpilibsuite/wpilib

closed 2020-04-16 by Starlight220