wpilibsuite/allwpilib #2485: How costly (runtime) is "transforming" a trajectory?

Starlight220, 2020-04-15:
It is documented that generating trajectories should be done in robotInit() as it is a costly operation. 
However, how heavy an operation is transforming/shifting/moving a trajectory? Will it affect robot performance if it is done in the middle of a match?

calcmogul, 2020-04-15:
Each trajectory only takes a millisecond or two to generate and is an O(n^2) operation on the length of the trajectory. You could conceivably generate them on-the-fly. We just recommend generating them on startup if possible because there's no real-time requirements at that point. Transforming the trajectories is an O(n) operation, so it should be negligible.

Starlight220, 2020-04-15:
Ok, great! And on more of a strategic question, how good of an idea would it be to use path "snippets" in teleop?

calcmogul, 2020-04-15:
Depends on the game challenge, how accurate your odometry is, and how accurate you need to be with game piece placement. 2019 was such a year.

Starlight220, 2020-04-15:
> 2019 was such a year

A year this would be effective, or a year this would utterly fail?

prateekma, 2020-04-15:
As Tyler said, the time it takes to transform the trajectory is negligible. 2019 is a good example of a year where on-the-fly trajectory generation could have been used to place hatch panels while being completely perpendicular to the velcro.

closed 2020-04-16 by Starlight220