wpilibsuite/allwpilib #2488: Add a PCM class?

Starlight220, 2020-04-19:
Why isn't there a Java one? I think I might have seen a C++ one.

AustinShalit, 2020-04-19:
Really, the compressor class is a sort of PCM class. The library is object oriented around actuators right now which means if we added a total PCM class, abstractions like solenoid and double solenoid might need to change. The Compressor class allows users to control the compressor and sensor functions of the PCM.

Starlight220, 2020-04-19:
Is there a way to read the PCM pressure switch?

AustinShalit, 2020-04-19:
You can use this method on Compressor:

https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/Compressor.html#getPressureSwitchValue()

Starlight220, 2020-04-19:
Ok, thanks. There isn't a way to get the pressure in the tanks from code, right?

AustinShalit, 2020-04-19:
>[The pressure switch] only has two values: open (yes, full pressure) and closed (low pressure). If you want to get the pressure, you’ll need a pressure transducer. AndyMark used to sell one, but I don’t see one right now. You can get one from Amazon or most of the usual parts places like McMaster or Mouser or even Amazon

[GeeTwo](https://www.chiefdelphi.com/t/pneumatics-programming-help/371856/41?u=austinshalit)

AustinShalit, 2020-04-19:
I've opened https://github.com/wpilibsuite/frc-docs/issues/627 to add this to the documentation.

Starlight220, 2020-04-21:
I think this has been settled.