wpilibsuite/allwpilib #2507: Why does it take more time to generate a trajectory for the first time (comparing to following generations).

OmerWexler, 2020-05-05:
Hey, I'm from the software team of BumbleB (3339).
Lately, we tested the capabilities of the TrajectoryGenerator.java and found out that when you generate a trajectory for the first time (since the robot booted) the generation time is significantly longer than what it was if you were to generate the same trajectory for the second, third, fourth, etc...

For example, we have found that generating a 3-meter straight trajectory (specifically quintic generation type) takes 418ms when first generated, 15ms for the second time, 16 for the third, and 4ms for the fourth time.
We have a few more examples where this effect persists if you need some more data.

My question is what could be the reason for that effect? And are there any workarounds?
Note that the measurements we made include only the specific function call that generates the trajectory.
The TrajectoryGenerator class we used is [here](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/trajectory/TrajectoryGenerator.java).

Thanks ahead.

ThadHouse, 2020-05-05:
A profiler would have to be hooked up to confirm, but the first time a piece of code is executed in Java, that code has to actually be compiled from Java Bytecode to actual Arm instructions. This can actually take a decent amount of time to perform, especially with complicated code like exists in the trajectory generator code. This is engrained in how Java actually work, and the way around it is to call it once during robotInit so that it doesn't take longer in autonomousInit. That's about all that can be done because of how the runtime and language actually work.

OmerWexler, 2020-05-06:
Ok, we will implement it so we have an initial call before the actual generation.
Thanks!

closed 2020-05-06 by OmerWexler