wpilibsuite/allwpilib #2508: Split units.h into separate headers for each unit

calcmogul, 2020-05-05:
Compile times drastically increase when it's included, and usually only a few unit types are used (meters, meters per second, radians, seconds). Splitting the header will better follow the include-what-you-use principle.

calcmogul, 2020-05-05:
This should be fairly straightforward copy-paste. Put the base macros in a units/base.h header and have units/meters.h, units/seconds.h, etc. for instantiations. Then instead of using units/units.h in library code, use the headers for the specific units actually used.

auscompgeek, 2020-05-06:
This potentially sounds like a good change to contribute upstream as well.

ThadHouse, 2020-05-06:
Most upstream places won't actually accept PR's like this would be. They actually prefer having a single header to make installation easier. Could be tried, but actually unlikely to get anywhere.

virtuald, 2020-05-06:
Might be worth making an issue on their repo and see if there's any interest.

Javascript has similar types of problems -- often the development is done across multiple files, but for many users it's more convenient to have a single file that can be copied into their source tree. They often use a program (webpack, other) to create the single-file version from the development source code. Perhaps a python script could be used to do a similar thing for C++ headers.

calcmogul, 2020-05-06:
Maybe we could convince them just how bad having all of this in a single header is by showing our compilation profiling.

Starlight220, 2020-05-07:
is it possible to have different headers, and then one that simply includes all of them? Then teams can include only what they need, but vendors/teams that use everything can have everything in a single statement. Even if it is possible, will it reduce compile time?

auscompgeek, 2020-05-07:
> is it possible to have different headers, and then one that simply includes all of them?

That would be the idea, yes.

> Even if it is possible, will it reduce compile time?

Continuing to include everything would not reduce compile times, no.

Starlight220, 2020-05-07:
> > Even if it is possible, will it reduce compile time?
> 
> Continuing to include everything would not reduce compile times, no.

It would only affect compile times if it is included, correct?

 If so, I think we should split and have an `allunits.h` header that includes everything. I can create the PR myself if we do this. Maybe find a better name for the "everything" header.

prateekma, 2020-05-07:
The all units header should remain `units/units.h` as it is now to not break teams' code that uses the units library. It should be deprecated though.

Starlight220, 2020-05-07:
So are we adding an "include everything" header under a different name and
eventually removing the current one, or is the units/units.h deprecation
simply to indicate that team should use the measurement-specific headers?

prateekma, 2020-05-07:
The new "include everything" header should be a depcreated `units/units.h` header. The deprecation message should indicate that users should use the measurement-specific headers.

Starlight220, 2020-05-07:
Is there a way to suppress cpp deprecation error messages in cases that need everything?
I will open a PR for this soon.

calcmogul, 2020-05-07:
`units/units.h` would be deprecated, but not for removal.

> Then teams can include only what they need, but vendors/teams that use everything can have everything in a single statement.

Vendors could use it if they wanted, but they really shouldn't. It's a bit silly to include units.h and pull in all 20+ units definitions (when are they ever going to use density or volume?) if all they need is meters and seconds.

We profiled the WPILib build, and parsing units.h took up most of the time by far. `units.h` turned into several megabytes of data the compiler has to churn through. We got rid of WPILib.h for a similar reason.

calcmogul, 2020-05-07:
> Is there a way to suppress cpp deprecation error messages in cases that need everything?

There are no cases that need everything. Here's the full list of unit types:

* acceleration
* angle
* angular_velocity
* area
* capacitance
* charge
* concentration
* conductance
* data
* data_transfer_rate
* density
* dimensionless
* energy
* force
* frequency
* frequency
* illuminance
* impedance
* inductance
* length
* luminous_flux
* luminous_intensity
* magnetic_field_strength
* magnetic_flux
* mass
* power
* pressure
* radiation
* solid_angle
* temperature
* time
* torque
* velocity
* voltage
* volume

Starlight220, 2020-05-07:
I wasn't aware there were so many types... And you are right. There is absolutely nothing that will need everything.

Starlight220, 2020-05-07:
It is much more work than I thought. I will work on it, but it might take a while.

prateekma, 2020-05-07:
Alright, let me know if you need any help.

Starlight220, 2020-05-09:
I may have messed up, it doesn't compile. I thought it would be much simpler, but it seems I was wrong. @prateekma, are you willing to take over? If so, do you want to try to salvage what I did or to start from scratch?

prateekma, 2020-05-09:
I'll see what I can do, but it will probably be after AP exams this week.

auscompgeek, 2020-06-27:
As it turns out upstream's v3.x branch already has split headers: https://github.com/nholthaus/units/pull/171

PR #2551 (2020-06-30): [wpiutil] Split units.h into separate headers for each unit
Closes #2508.
commit d30d1088 (2020-06-30): [wpiutil] Split units.h into separate headers for each unit (#2551)

Closes #2508.
closed 2020-06-30 by PeterJohnson