wpilibsuite/allwpilib #2509: [command] Refactor PerpetualCommand structure

Starlight220, 2020-05-09:
After viewing multiple issues with the current `PerpetualCommand` structure, including #2302, #2389; and some discussion on the wpilib Gitter, it would seem that the PerpetualCommand needs some refactoring. The main problem is that the current implementation doesn't accommodate command types with irregular flow (`CommandGroup`, `InstantCommand`).

At first, there were a few ideas I thought of, but each has its problem:
1. **Overriding the decorators in the different command types** : The only advantage the decorators should have is cleaner syntax (there should be no performance diff between using a decorator v.s. using the regular constructor). This would violate that. 
2. **Adding a Constructor overload for each different command type**: This wouldn't be consistent, as it would be easy to accidentally use the regular overload with a different command type,  as (at least in Java) overloads are decided at compile-time. Polymorphism checks can't be used, as (at least in Java), constructor delegation must be the first statement in the constructor. It might also result in a less clean syntax. 

Therefore, I have thought of a solution that has none of the above problems (perhaps except the syntax one). It has the following parts: 

- [ ] `RunCommand` should extend `PerpetualCommand`, so it can be used to perpetuate lambdas/`InstantCommands `while giving the best possible type data (the methods can return `PerpetualCommand` instead of the polymorphic `Command`). 
- [ ] Having **factory methods** (probably in `PerpetualCommand`) with overloads + polymorphic checks. This avoids the problems in _idea 2 (constructor overloads)_. 
- [ ] The current public constructor in PerpetualCommand should be deprecated, then made protected after a year (as it's a public API change).

I am pretty sure there are no Java problems with this, and I doubt there are any C++ problems (can someone make sure I'm right? I am not a C++ programmer) either. I would be happy to implement this in Java. If this is implemented, it would fix the two issues mentioned above. 

@PeterJohnson, @AustinShalit, @calcmogul - your thoughts?

Starlight220, 2020-06-07:
@Oblarg, should I start working on this, or to wait for you to review? 
The 2021 season is approaching, and there are multiple stuff to do with this issue other than the change itself - adding to frc-docs, perhaps creating an example, etc.

Oblarg, 2020-06-07:
I agree that `RunCommand` should extend `PerpetualCommand`.

I do not think the constructor nor the decorator should be deprecated; they perform a clearly-defined function and that function happens to not work with commands that are written a certain way.  The task of repeating a command whose end condition cannot simply be removed should be handled by a different type entirely (maybe `RepeatCommand`?), with its own decorator.

Oblarg, 2020-06-07:
I think they very much should care - restarting a command repeatedly versus removing the end condition are fundamentally different things, and there are a number of commands that work with both and for which both are useful for different things.  This isn't a case of a single abstraction with different implementation details - these are two different abstractions entirely, that solve different problems.

I think it will be reasonably clear, if a `RepeatCommand` exists, what the difference between the two is.

Starlight220, 2020-06-08:
So users are supposed to call `.perpetually()` for some command types and `.repeatedly()` for others? 
This seems like a major footgun.

carbotaniuman, 2020-06-27:
I think the proper way of thinking about this is that there should a reschedule() method of some sort that allows commands to customize how they want to be rescheduled.

There should be no way to perpetually run an InstantCommand, if you want that to happen than use a RunCommand instead (or rewrite InstantCommand to be a RunCommand internally).

Rescheduling command groups could be established by just readding everything, given a clear enough documentation of the semantics.

Starlight220, 2020-06-27:
> There should be no way to perpetually run an InstantCommand, if you want that to happen than use a RunCommand instead (or rewrite InstantCommand to be a RunCommand internally).

I agree with this, but you have to support `InstantCommand.perpetually()` unless you want to refactor the whole setup. I think that it should return a `RunCommand`, together with documenting the correct way to do it. The same goes for other command types (groups etc).

The way I see the discussion here, it's a matter of "what does running a command perpetually mean?":

- As I understand it, @Oblarg thinks that it should mean removal of the end condition - other ways of repetition (rescheduling, etc.) should be done via a different API.
- My thoughts are that it should mean running the command constantly, however logically possible (while documenting side effects, etc.). That solution I am proposing makes that possible.
- @carbotaniuman - I haven't fully understood what you are proposing/your thoughts here, more details will be welcomed. 

Having the functionality of building a `PerpetualCommand` in a method that is overridden by command types is the more OOP way of doing this, but then there are two options - overriding `.perpetually()`, or another method.
 - Some of the WPI team has expressed their opinions that the functionality should not be in the decorators. Having `.perpetually()` overridden will be inconsistent with this. 
- If in a different method, what will be the responsibility of this method, and how will teams implement it. This will also add to the (already relatively long) list of methods that teams should implement in custom command classes. It also (probably) needs a more exposed API, and won't help a lot. 

**tl;dr** I think that custom perpetuating should be done by overriding `.perpetually()` with functionality, but is inconsistent with WPI team decisions/opinions. Factory methods are a secondary option, but limits custom team implementations.

carbotaniuman, 2020-07-13:
Hmmmm. When you think about it, there's mainly two ways you can perpetually run a command, either rescheduling it, by ignoring the end conditions.

This of course still doesn't really fit InstantCommand (maybe have it act similar to RunCommand?), but it would at least allow CommandGroups to be perpetually run.

The exact implementation could be an Enum that all commands return, or we could lift perpetually from a decorator to a standard method.

Starlight220, 2020-07-13:
I think `InstantCommand.perpetually()` is bad/non-idiomatic design (by the team), with lack of knowledge about `RunCommand` to blame. Since we must support it, I suggest for the internal implementation to "correct" the bad design - ie changing it to a `RunCommand`. 
CommandGroups should have a way of perpetuation - logically, this would be rescheduling the group's contents. 

I see two API/implementation options: 
1. initialization of `PerpetualCommand`s via factory methods instead of constructors
2. having the functionality in `.perpetually()` (rename the method if needed), then each command type that needs a different perpetuation can override and provide custom functionality (sometimes resulting in custom PerpetualCommand subclasses). Move the current PerpetualCommand functionality into an EndlessCommand class that extends the abstract? PerpetualCommand class. Then, we can have different, well-documented (because of the name difference) types of PerpetualCommands: EndlessCommand (what we have currently - removes end condition), RunCommand (for lambdas/InstantCommands), RescheduleCommand (for groups - reschedules contents). In extreme cases, teams can define custom PerpetualCommand subclasses.

Starlight220, 2022-04-21:
RepeatCommand (added in #4009) solves most of the issues here.

#4176 would clean up any naming confusion issues.

closed 2022-04-21 by Starlight220