wpilibsuite/allwpilib #2514: Add a way to get CAN bus timestamps for motor controller measurements

calculusrobotics, 2020-05-19:
It would be nice to have a way to get a timestamp on motor controller status frames, particularly encoder data (position and velocity). This way we could tell how old a measurement is and properly account for it in controllers and estimators on the RoboRIO.

calculusrobotics, 2020-05-19:
Also if this is possible, would the timestamps be from when the frame got sent or received through the CAN bus? Thanks!

ThadHouse, 2020-05-19:
It would be when the frame was received by the Rio. Its not realistically possible to get a timestamp from the send.

ThadHouse, 2021-05-01:
This isn't actually something we can do on our end. We expose the functionality to vendors, as there is a timestamp on the CAN messages, however its up to them to actually give those values to the user. So theres nothing we can do to fix this.

closed 2021-05-01 by ThadHouse