wpilibsuite/allwpilib #2518: Add a simple File Server to wpiutil

ThadHouse, 2020-05-22:
In a similar vain to the port forwarder, a simple uv based static file server could be super useful, especially with more websocket supporting protocols. 

In wpiutil, this could be fairly easy to run from both C++ and Java.

virtuald, 2020-07-06:
It would be great if this could be integrated with a websocket. 

I have a terrible version of this derived from the vision pi stuff @ https://github.com/virtuald/halsim-websocket/blob/master/src/main/native/cpp/HALSimHttpConnection.cpp , but as you've noted it won't work on Windows.

PeterJohnson, 2020-09-14:
The HALSIM websocket server extension has this, but of course that's not usable from Rio robot code.

PR #7527 (2024-12-14): [wpinet] Add simple web server
Example use:
```java
WebServer.start(5800, Filesystem.getDeployDirectory());
```

TODO:
- [x] Generate pages for directory browsing
- [x] ~~Support PUT and DELETE operations~~ (deferred)
- [x] Fix Windows

Closes #2518
closed 2024-12-14 by PeterJohnson