wpilibsuite/allwpilib #2521: CommandScheduler singleton

ItayZiv, 2020-05-31:
Would it be possible to have `CommandScheduler` have a singleton instance but also be able to be created sepertly (additional instances)? 
I can see a use case of people want to use command-based while running another thread/notifer. (Either with a public ctor or a "factory")

Starlight220, 2020-05-31:
You mean something similar to the NTI setup - you have the default instance, but can also create your own for unit tests, etc?

calcmogul, 2020-05-31:
Command-based was designed for cooperative multitasking in a single thread. Separate instances in a thread-per-subsystem kind of thing may work if they never talk to each other. If you have to coordinate actions between them, you're going to have thread safety issues.

ItayZiv, 2020-06-02:
Yeah, this would just be a advanced thing (Thats why I thought maybe more of a factory method would be more fit). Any thread saftey would be something to be handled by the users (Subsystem methods, or just seperate subsystems).
This could be nice (with some docs on how to use that and notifiers for running stuff like PID on a "fast loop" (200+ hz) and some more complex stuff like, trajectory generation, etc... on a "slow loop")
Currently the only real alternative would be to do messy stuff with each command/subsystem and use the 1 CommandScheduler or just only have on of the threads on commad-based

Starlight220, 2020-06-02:
Would running a specific command at a higher rate solve this? If so, it theoretically can be done internally with a Notifier, via the main instance.

ItayZiv, 2020-06-02:
Yea, that's just much harder, and requires some extra thread safety, and if that's just built in then subsystems would need to be thread safe...

Starlight220, 2020-06-02:
Why? If the commands obey the requirement system, then each subsystem should only be "messed with" by one thread.

Starlight220, 2020-09-14:
@ItayZiv is `NotifierCommand` enough for what you thought of?
I don't think there's a reason to add the option for multiple `CommandScheduler` instances - too much mess, and many things (such as requirements, etc) would have to be static. And teams would still mess up with multithreading/thread-safety. 
Teams that would have enough knowledge to run commands in parallel threads _safely_ would also be able to handle this themselves, possibly using `NotifierCommand`.

calcmogul, 2021-08-24:
The "controller running at a faster rate" use-case has now been addressed by `TimedRobot.addPeriodic()`. Instead of separate `CommandScheduler` instances, I suggest the following architecture:

1. Run the decision making and state machine transitions (`CommandScheduler`'s job) at 50Hz in `teleopPeriodic()`.
2. Run the controller output updates in subsystem periodic functions at 100+Hz via `TimedRobot.addPeriodic()`. Note this would be a separate function from `edu.wpi.first.wpilibj2.command.Subsystem.periodic()` since that would only run at 50Hz.

This necessitates the controller living in the subsystem instead of the command, but I prefer that; commands should be thin wrappers that tell the subsystem what to do then wait for the subsystem to finish.

closed 2021-08-24 by calcmogul