wpilibsuite/allwpilib #2522: Add a neat/clean syntax for stacking Filters, Controllers, etc.

Starlight220, 2020-05-31:
In some cases, a PIDController, FeedForward, and a Filter are all used simultaneously. This can cause some unpleasant syntax, such as `motor.set(feedforward.calculate(pidcontroller.calculate(filter.calculate(raw_input))))`. I suggest an API that will be used similarly to: 
(In the construction)
``` java
LinearFilter filter = ...
PIDController pidcontroller = ...
FeedForward feedforward = ...
DoubleProccess pipeline = new DoubleProcess(filter, pidcontroller, feedforward);
```
And at the use site: 
```java
motor.set(pipeline.calculate(raw_input));
```

It should also support custom "stages", and perhaps commonly used "stages" such as accounting for gearing can be provided. 
As I see this, this can either be implemented either by an interface that PIDController, Filters, FeedForward, etc will implement, or simply use a stdlib function type (in java, it can be [`DoubleUnaryOperator`](https://docs.oracle.com/javase/8/docs/api/java/util/function/DoubleUnaryOperator.html), idk about c++).

calcmogul, 2020-05-31:
The pipeline you suggested there isn't actually a linear pipeline. `filter` provides the measurements for `pidcontroller`, but `pidcontroller` and `feedforward` are separate branches in the block diagram with a summation block after them.

The following is how I'd write a basic flywheel subsystem with a controller. It expresses its intent well, although I'm open to ways to simplify common use cases.

```java
public class Flywheel {
  private Encoder encoder = new Encoder(1, 2);
  private VictorSPX motor = new VictorSPX(0);

  private LinearFilter filter = LinearFilter.singlePoleIIR(0.2);
  private PIDController pidController = new PIDController(kP, kI, kD, 0.02);
  private SimpleMotorFeedforward feedforward = new SimpleMotorFeedforward(kV, kA);

  public double controller(double measurement, double setpoint) {
    double filteredMeasurement = filter.calculate(measurement);
    return pidController.calculate(filteredMeasurement, setpoint) + feedforward.calculate(setpoint, 0);
  }

  public void periodic() {
	motor.set(controller(encoder.getRate());
  }
}
```

We've tried higher-order function compositions in the past (https://github.com/wpilibsuite/allwpilib/pull/544). It was overabstracted and had a higher learning curve than the example above. That's why we went with the "calculate()" API instead and let users abstract things how they saw fit.

ItayZiv, 2020-06-02:
The only thing I could think of is having a `FilteredPIDController` that self-contains a filter that it runs on measurments (user specifies passed a filter in the ctor).
Not sure how useful that is. (Another thing is that `ProfiledPIDController` could take in a `PIDController`, which could be a `FilteredPIDController` since it can extend `PIDController`.

Starlight220, 2020-06-02:
@calcmogul, what I am suggesting is have the `controller` method from the code snippet you included, in the library. I think teams can also benefit from adding other "stages" to it: accounting for gearing, clamping, etc.

ItayZiv, 2020-06-02:
Also, this can also just be done with a lambda (`DoubleFunciton` in Java)

calcmogul, 2020-06-02:
> The only thing I could think of is having a FilteredPIDController that self-contains a filter that it runs on measurements (user specifies passed a filter in the ctor).

We made ProfiledPIDController because using TrapezoidProfile together with PIDController has essential complexity we can abstract away. The filter API, on the other hand, is basically impossible to use incorrectly. Abstracting away a single trivial composition doesn't seem worth it from a library or user point of view.

> Another thing is that ProfiledPIDController could take in a PIDController, which could be a FilteredPIDController since it can extend PIDController.

That seems like excessive abstraction. The current API with the PIDController ctor arguments plus trapezoid profile constraints is easier to use.

> Also, this can also just be done with a lambda (`DoubleFunction` in Java)

Whether users make their own function and pass the value into PIDController or take a lambda in the API is a question of where you want the plumbing to be invoked. The user has to specify the plumbing either way, so we decided to stay away from higher-order functions to simplify library internals. Each component has a job, and how it's plumbed together with other components is a higher level concern the component itself shouldn't worry about. We use higher-order functions in command-based because those objects are generally combining a bunch of pieces at once in a nontrivial way.

> In theory, this concept can be extended to a single object that does PID, FF and filtering all in one, but not sure if that's even desired.

PIDController used to be this way prior to 2019. We moved away from that to get rid of combinatorial explosions of default arguments and optional features (e.g., having a class for PID, PID + filtering, PID + FF + filtering). It made the internals less convoluted and easier to maintain. It also better follows "don't pay for what you don't use". We have a lot of decoupled components now, and letting the user decide how to plumb them together is a feature as long as how they should do so is obvious.

ItayZiv, 2020-06-02:
Yea I agree, just saying thats closest it could get, but yeah I 100% agree

closed 2020-08-20 by Starlight220