wpilibsuite/allwpilib #2528: Break long dependency chains

PeterJohnson, 2020-06-12:
Currently there are some long dependency chains in the code base that make it hard to use code separately without copy and pasting.  Some examples:
- Commands -> Sendable -> SendableBuilder -> NetworkTables
- Many classes -> LiveWindow/Sendable -> NetworkTables
- Many classes -> HAL Usage reporting

It should be possible to break these dependency chains via more separated or stacked interfaces and/or DI.

Starlight220, 2020-07-02:
> * Commands -> Sendable -> SendableBuilder -> NetworkTables

Wasn't this the whole point in `Command`/`CommandBase`?

> * Many classes -> HAL Usage reporting

If you mean the source of usage stats published at the end of the season, then it maybe can be extracted to a `UsageReported` interface. But that depends on which codebase level the usage reporting is. 

> * Many classes -> LiveWindow/Sendable -> NetworkTables

I doubt that there is anything that can be done about this, as it is already relatively DI. Perhaps replace `void initSendable(Builder)` with `Map<T, Map.Entry<Supplier<T>, Consumer<T>>>  reportProperties()` that returns a map of the properties, getters, setters. Maybe rework the LW/Sendable -> NT connection somehow.

PeterJohnson, 2020-09-14:
We've started to break the HAL Usage Reporting chain, for example in wpimath.