wpilibsuite/allwpilib #2535: Drivetrain kinematics inheritance feature.

OmerWexler, 2020-06-16:
Hey, 
We encountered a problem when developing autonomous trajectories using the TrajectoryConfig class. We are trying to use the most generic code to generate trajectories for both differential and swerve drives, during development we noticed the different kinematics classes (for swerve and differential) are completely unrelated (inheritance wise) and so we had to write very repetitive code to handle both classes for the trajectory generation (i.e having to write to identical methods with the only difference being the type of the kinematics parameter). 

We would like you to consider adding a base kinematics class of which all advanced kinematics classes will be derived (general example: the SwerveDriveKinematics class will inherit from a generic abstract kinematics class).

We have considered trying to write our code in such a way that supports both classes, although we have reached a conclusion that our code will only get more and more repetitive and advanced without having both kinematics types inherit from a common parent.

Thanks ahead, BumbleB #3339.

calcmogul, 2020-06-16:
What do you visualize the common interface would look like? Most of the function signatures between the differential drive and swerve drive kinematics classes are different, so I'm having trouble seeing what the common interface would be.

ItayZiv, 2020-06-19:
The only way that would work would be to do some weird generification that would make the classes just not easy to use, and would probably still need code duplication on each case to get anything useful from that.

OmerWexler, 2020-06-20:
Ok, we will try working our way around. The difference between the classes does sound like a good reason to keep them separate.
Thanks.

closed 2020-06-20 by OmerWexler