wpilibsuite/allwpilib #2550: Consider using multimedia timers for simulation on Windows

PeterJohnson, 2020-06-27:
The current approach of waiting on a timed mutex may not always have the best timing.  Using the Multimedia Timer API might give more consistent results.  https://docs.microsoft.com/en-us/windows/win32/multimedia/multimedia-timer-reference

PeterJohnson, 2020-06-27:
We might just be able to force bump up the timer precision and keep doing what we're doing currently.

TheTripleV, 2020-06-27:
Would helper functions for this go in WindowsSupport.h? And, does the sim have init/exit funcs where that header could be included?

PeterJohnson, 2020-06-28:
The initialize stuff goes in HAL_Initialize in hal/src/main/native/sim/HAL.cpp.  There's not really a HAL_Exit, but you could install an atexit handler?  Unless the code is huge, I would just put it directly in HAL.cpp with an #ifdef _WIN32.

PR #2557 (2020-06-29): [sim] Set Windows timer precision to 1ms
Closes #2550 .
closed 2020-06-29 by PeterJohnson