wpilibsuite/allwpilib #2574: Add websocket HALSIM extension

virtuald, 2020-07-06:
I've got an initial lightly tested version here and an accompanying demo: https://github.com/virtuald/halsim-websocket/

It's just a server now, but Peter wants it to have a client socket also. Sounds like work.

TheTripleV, 2020-07-11:
I'm thinking of taking a crack at this using imgui-js. Are there other/better libraries that should be considered?

virtuald, 2020-07-12:
Heh, sounds like an interesting library. That might be an effective approach.

@amorygalili is using some component library to create frc-web-components, he was planning on eventually creating a frontend for my websocket server (which was why I created the server in the first place). https://github.com/frc-web-components/frc-web-components is the repo -- I've been really impressed by the small pieces I've seen, but I don't know a ton about it.

amorygalili, 2020-07-12:
FRC Web Components uses the [LitElement](https://lit-element.polymer-project.org/) component library. I don't know much about imgui or the js bindings, but I think you could probably combine them if you want to. Right now FRC Web Components works with NetworkTables, but it wouldn't be too much effort to add support for the HALSIM extension. I was planning on working on this at some point, but if someone else wants to use it I can look into it this week.

zhiquanyeo, 2020-08-20:
Resolved by https://github.com/wpilibsuite/allwpilib/pull/2589

closed 2020-08-21 by AustinShalit