wpilibsuite/allwpilib #2578: Add JSON URL to Command JSON files

Starlight220, 2020-07-07:
If the JSONs are stored on the internet anyway, why isn't the URL in the JSON file so it can be downloaded with the "online" method? 
There are links in frc-docs for both files, but not a simple-text URL (workaround: copy link address).

ThadHouse, 2020-07-07:
Because they are guaranteed to be installed with the installer, they always show up as an offline install option assuming users are using the recommended method. So it just never was important. Go ahead and PR it thought.

Daltz333, 2020-09-20:
@ThadHouse What was the actual reason the URL was left out?

ThadHouse, 2020-09-20:
Because that url is only for updates, and they don't update. The only use for that is so the vscode auto updater can find it. But it doesn't need updates, so no need for url.

Daltz333, 2020-09-20:
@Starlight220 Are you fine with closing this issue than?

Starlight220, 2020-09-20:
I still prefer that the json have a URL (or at least that a URL for downloading it will exist and be accessible) so that it can be easily  added to the project without VS Code, or maybe even installing WPILib. This is *one* way to do it, if there are others I'll be happy to listen.

ThadHouse, 2020-09-20:
Adding the url to the Json doesn't solve that though, because how would a user find the Json in the first place? A link in the docs is the only thing we can do for that. Having the url in the Json is useless if a user can't find the Json file anyway.

Starlight220, 2020-09-20:
Oh well. Are the URLs the same every year? If so I can just write a script that adds the jsons I use (assuming the vendordep task GR PR is merged...).

ThadHouse, 2020-09-20:
The url is not currently fixed, and that's part of the problem as well. When we do the main branch rename old scripts will break.

Daltz333, 2020-09-20:
I think the solution would be to move the vendordep json to artifactory then?

Daltz333, 2020-09-20:
Additionally frc-docs link checking ensures we always get valid links

Starlight220, 2022-05-02:
The `vendordep` Gradle task removes the need for this; closing.

closed 2022-05-02 by Starlight220